Seeed-Studio/Grove_Gesture #5: Transmission Error

Pasadenasman, 2020-12-08:
Hi guys, 

I'm working around that PAJ7620U2 sensor and i'm stuck with this transmission error i got. It seems the init processus isn't ok and can't figure what is the problem. 
I'm using an esp32 board (lilygo TTGO T-Display) with the default I2C pin (pin 21 & 22) as SDA and SCL.

I'm using this code and the libraries connected to it. 

`/*
 * Copyright (c) 2015 seeed technology inc.
 * Website    : www.seeed.cc
 * Author     : Wuruibin
 * Modified Time: June 2015
 * Description: This demo can recognize 9 gestures and output the result, including move up, move down, move left, move right,
 * 				move forward, move backward, circle-clockwise, circle-counter clockwise, and wave.
 * 
 * The MIT License (MIT)
 *
 * Permission is hereby granted, free of charge, to any person obtaining a copy
 * of this software and associated documentation files (the "Software"), to deal
 * in the Software without restriction, including without limitation the rights
 * to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
 * copies of the Software, and to permit persons to whom the Software is
 * furnished to do so, subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in
 * all copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
 * FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
 * AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 * LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
 * OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
 * THE SOFTWARE.
 */

#include <Wire.h>
#include "paj7620.h"

/* 
Notice: When you want to recognize the Forward/Backward gestures, your gestures' reaction time must less than GES_ENTRY_TIME(0.8s). 
        You also can adjust the reaction time according to the actual circumstance.
*/
#define GES_REACTION_TIME		500				// You can adjust the reaction time according to the actual circumstance.
#define GES_ENTRY_TIME			800				// When you want to recognize the Forward/Backward gestures, your gestures' reaction time must less than GES_ENTRY_TIME(0.8s). 
#define GES_QUIT_TIME			1000

void setup()
{
	uint8_t error = 0;

	Serial.begin(115200);
	Serial.println("\nPAJ7620U2 TEST DEMO: Recognize 9 gestures.");

	error = paj7620Init();			// initialize Paj7620 registers
	if (error) 
	{
		Serial.print("INIT ERROR,CODE:");
		Serial.println(error);
	}
	else
	{
		Serial.println("INIT OK");
	}
	Serial.println("Please input your gestures:\n");
}

void loop()
{
	uint8_t data = 0, data1 = 0, error;
	
	error = paj7620ReadReg(0x43, 1, &data);				// Read Bank_0_Reg_0x43/0x44 for gesture result.
	if (!error) 
	{
		switch (data) 									// When different gestures be detected, the variable 'data' will be set to different values by paj7620ReadReg(0x43, 1, &data).
		{
			case GES_RIGHT_FLAG:
				delay(GES_ENTRY_TIME);
				paj7620ReadReg(0x43, 1, &data);
				if(data == GES_FORWARD_FLAG) 
				{
					Serial.println("Forward");
					delay(GES_QUIT_TIME);
				}
				else if(data == GES_BACKWARD_FLAG) 
				{
					Serial.println("Backward");
					delay(GES_QUIT_TIME);
				}
				else
				{
					Serial.println("Right");
				}          
				break;
			case GES_LEFT_FLAG: 
				delay(GES_ENTRY_TIME);
				paj7620ReadReg(0x43, 1, &data);
				if(data == GES_FORWARD_FLAG) 
				{
					Serial.println("Forward");
					delay(GES_QUIT_TIME);
				}
				else if(data == GES_BACKWARD_FLAG) 
				{
					Serial.println("Backward");
					delay(GES_QUIT_TIME);
				}
				else
				{
					Serial.println("Left");
				}          
				break;
			case GES_UP_FLAG:
				delay(GES_ENTRY_TIME);
				paj7620ReadReg(0x43, 1, &data);
				if(data == GES_FORWARD_FLAG) 
				{
					Serial.println("Forward");
					delay(GES_QUIT_TIME);
				}
				else if(data == GES_BACKWARD_FLAG) 
				{
					Serial.println("Backward");
					delay(GES_QUIT_TIME);
				}
				else
				{
					Serial.println("Up");
				}          
				break;
			case GES_DOWN_FLAG:
				delay(GES_ENTRY_TIME);
				paj7620ReadReg(0x43, 1, &data);
				if(data == GES_FORWARD_FLAG) 
				{
					Serial.println("Forward");
					delay(GES_QUIT_TIME);
				}
				else if(data == GES_BACKWARD_FLAG) 
				{
					Serial.println("Backward");
					delay(GES_QUIT_TIME);
				}
				else
				{
					Serial.println("Down");
				}          
				break;
			case GES_FORWARD_FLAG:
				Serial.println("Forward");
				delay(GES_QUIT_TIME);
				break;
			case GES_BACKWARD_FLAG:		  
				Serial.println("Backward");
				delay(GES_QUIT_TIME);
				break;
			case GES_CLOCKWISE_FLAG:
				Serial.println("Clockwise");
				break;
			case GES_COUNT_CLOCKWISE_FLAG:
				Serial.println("anti-clockwise");
				break;  
			default:
				paj7620ReadReg(0x44, 1, &data1);
				if (data1 == GES_WAVE_FLAG) 
				{
					Serial.println("wave");
				}
				break;
		}
	}
	delay(100);
}
`

I got this from the serial port :
`10:24:49.077 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
10:24:49.077 -> configsip: 0, SPIWP:0xee
10:24:49.077 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
10:24:49.077 -> mode:DIO, clock div:1
10:24:49.077 -> load:0x3fff0018,len:4
10:24:49.077 -> load:0x3fff001c,len:1216
10:24:49.077 -> ho 0 tail 12 room 4
10:24:49.077 -> load:0x40078000,len:9720
10:24:49.077 -> ho 0 tail 12 room 4
10:24:49.077 -> load:0x40080400,len:6352
10:24:49.077 -> entry 0x400806b8
10:24:49.216 -> 
10:24:49.216 -> PAJ7620U2 TEST DEMO: Recognize 9 gestures.
10:24:49.216 -> INIT SENSOR...
10:24:49.216 -> Addr0 =20,  Addr1 =76
10:24:49.216 -> wake-up finish.
10:24:49.216 -> Transmission error!!!
10:24:49.216 -> Transmission error!!!
10:24:49.216 -> Transmission error!!!
10:24:49.216 -> Transmission error!!!
10:24:49.355 -> Transmission error!!!
10:24:49.355 -> Transmission error!!!
10:24:49.355 -> Transmission error!!!
10:24:49.447 -> Transmission error!!!
10:24:49.494 -> Transmission error!!!
10:24:49.494 -> Transmission error!!!
10:24:49.632 -> Transmission error!!!
10:24:49.632 -> Transmission error!!!
10:24:49.632 -> Transmission error!!!
10:24:49.724 -> Transmission error!!!
10:24:49.724 -> Transmission error!!!
10:24:49.724 -> Paj7620 initialize register finished.
10:24:49.724 -> INIT OK
10:24:49.724 -> Please input your gestures:
10:24:49.724 -> 
10:24:49.724 -> Transmission error!!!
10:24:49.909 -> Transmission error!!!
10:24:50.323 -> anti-clockwise
10:24:51.153 -> Transmission error!!!
10:24:53.366 -> anti-clockwise
10:24:54.381 -> Transmission error!!!
10:24:57.384 -> anti-clockwise
10:24:57.476 -> Transmission error!!!
10:24:58.491 -> Transmission error!!!
10:24:59.784 -> anti-clockwise
10:25:00.754 -> Transmission error!!!
10:25:04.398 -> Transmission error!!!
10:25:04.491 -> Transmission error!!!
10:25:05.505 -> Transmission error!!!
10:25:07.997 -> Transmission error!!!
10:25:09.611 -> Transmission error!!!
10:25:09.936 -> Transmission error!!!
10:25:10.673 -> Transmission error!!!
10:25:11.227 -> Transmission error!!!
10:25:11.412 -> Transmission error!!!
10:25:11.642 -> Transmission error!!!
10:25:11.735 -> Transmission error!!!
10:25:15.983 -> Transmission error!!!
10:25:16.167 -> Transmission error!!!
10:25:19.308 -> anti-clockwise
10:25:19.815 -> Transmission error!!!
10:25:22.031 -> Transmission error!!!
10:25:22.123 -> Transmission error!!!
10:25:23.827 -> Transmission error!!!
10:25:24.011 -> Transmission error!!!
10:25:24.519 -> Transmission error!!!
`
i'm not moving in front of the sensor to generate the anti-clockwise move. Even while I'm hidding the sensor with my finger, the transmission error and anti-clockwise data are still looping.

Can somebody point out what's wrong with this ?

Pillar1989, 2020-12-09:
@Pasadenasman  I'm sorry, but I'm seeing garbled code all over the screen and I can't really read your question. Maybe you can clean up the problem and do a layout of the problem using markdown format.

uecken, 2020-12-25:
I use this sensor using M5StickC by Grove connector.
After writing from Arduino IDE, sensor is good state,
however restarting M5StickC, "INIT ERROR,CODE:2" was output and cannot use this sensor.

M5Stack is no problem.
Can anyone resolve this?

Lesords, 2024-10-16:
Hello,

I'm very sorry to have kept you waiting so long.

Does this problem still exist?If so, can you organise a description of your problem?

closed 2024-10-30 by Lesords